GlobalPlatform/psa-api #5: PSA Firmware  API  FWU: support on flash nand with ECC

jamike, 2022-11-04:
For platform having nand flash with ecc, the write access  need to be aligned on a minimum size which fluctuates according to platform.
Unaligned access on this plaform can be supported the FWU update service on device , but it adds complexity in this service and requires an additional block in flash to support unaligned access.  
Another solution is to  add an implementation define: PSA_FWU_WRITE_SIZE_UNIT 
On plaform without alignment constraint , the define is set to #define PSA_FWU_WRITE_SIZE_UNIT 1
for a plaform having 32 bytes aligment constraint, the value is set to #define PSA_FWU_WRITE_SIZE_UNIT 32
and the parameter   image_offset and  block_size from psa_fwu_write  are no more in bytes but in number of PSA_FWU_WRITE_SIZE_UNIT. 
 
NB: In Firmware_Update_API-1.0-bet.0, only one implementation define  (PSA_FWU_MAX_WRITE_SIZE)

athoelke, 2022-11-04:
This does look like a use case constraint that would be good to consider.

### Background

In 1.0 so far, we have not changed the specification around the `psa-fwu_write()` function that was present in 0.7. It would be very valuable to get feedback on whether this area should be more clearly specified.

The existing definition is very flexible, and provided no requirements on alignment, or ordering, of the data provided by the Update client. However, it also permits an implementation to have constraints (such as block alignment, or sequencing) that are not visible in the API - so the Update client has to just know what is expected.

I think (but am not sure) that the flexibility was to allow [hypothetical] devices with very limited bandwidth to present update blocks in the order they are received, without needing to temporarily store a block (or discard it) if received out of order.

However, I suspect that for the majority of implementations, it is most efficient for the service to receive blocks that are correctly aligned, in order, and preferably as large as possible (`PSA_FWU_MAX_WRITE_SIZE`). And ideally, the implementation would like to require that the Update client provides the data in this way, as this allows these assumptions to be designed-in to the implementation.

### API ideas

Regarding the API, I am not sure that changing the unit size for the API is going to be intuitive. This could lead to errors when a developer forgets to divide the buffer size by `PSA_FWU_WRITE_SIZE_UNIT`.

I think that a perhaps 'granule size' or 'alignment size' type of implementation-defined constant might be better - and permit an implementation to report an INVALID_ARGUMENT error if the alignment is not correct:

* `(image_offset & (PSA_FWU_WRITE_ALIGN-1)) == 0`
* `(block_size & (PSA_FWU_WRITE_ALIGN-1)) == 0` - or perhaps zero-pad any incomplete alignment block?

If alignment is not required then this value could be 1, the largest value would be `PSA_FWU_MAX_WRITE_SIZE`. In all cases the maximum write size must be aligned: `PSA_FWU_MAX_WRITE_SIZE % PSA_FWU_WRITE_ALIGN == 0`.

I am also wondering about exposing a 'sequential writes' requirement, perhaps via a flag in the component info:

* If this flag is set, then the Update client must write the image in sequence.
* With the alignment constraint, this _could_ also require that a non-final write must be fully aligned; but the final write could end with an incomplete alignment block?

The specification could then recommend that the Update client writes the image in order, providing blocks of `PSA_FWU_MAX_WRITE_SIZE` until the final block: this approach will be guaranteed to work on any implementation. More flexible approaches could be permitted, but Update clients that exploit the flexibility will not work on all implementations.

jamike, 2022-11-04:
The APi Idea with PSA_FWU_WRITE_ALIGN is fine, even if it does not address the incomplete alignment block at start and end.

- Incomplete alignment block at start  can occur if the download area of one image is stick to area from another image without respecting alignment. (i.e for optimizing  flash usage especially if alignment constraint is large enough to justify such mapping )
- Incomplete alignment block at end  can occur at every update (i.e more space is reserved for futher update in the download area)
So the major drawback to consider is incomplete alignment block at end , especially for large alignment it will force to pass a padded buffer to the api. For the flash access  (operation contributing the most in term of timing)   on FWU service , it should remain the same because the complete alignment block need to be written anyway.

athoelke, 2022-11-15:
@jamike - sorry for the pause in responding - I have been working through the TF-M implementation of the API, and extracting some additional issues to address in the final 1.0.0 spec.

I would definitely like to find a way to address the alignment issue, at least to some extent, as it has a significant impact on systems that require ECC block-wise writes to the flash.

> * Incomplete alignment block at start can occur if the download area of one image is stick to area from another image without respecting alignment. (i.e for optimizing flash usage especially if alignment constraint is large enough to justify such mapping )

Exposing the internal offset of the start of a not-block-aligned image would make using this API significantly more complex. We would have to expose the image offset and the client would have to do a special first write to align the remaining writes.

I would like to suggest a simple initial approach (as described above) that only requires that `image_offset` is aligned, and specifying that a non-aligned `block_size` will result in impdef padding of the final alignment-block. A client should only do the latter for the last block in a file (as the implementation does not know how big the image is).

If alignment is provided by the calls from the client, then the writes would not have to be strictly sequential - although the client would have to track whether it has provided all of the data before calling `psa_fwu_finish()`.

For an implementation that does not support persistent staging (i.e. images in WRITING or CANDIDATE state are discarded on reboot &mdash; see #8), then specifying the alignment via this API is not essential, as the implementation can buffer partial blocks in RAM, and `psa_fwu_finish()` can complete the last block. This would depend on sequential writes, but that can be enforced via maintain additional volatile state about the writing progress.

I think the idea of a flag for 'requires sequential writes' needs more thought &mdash; requiring that an implementation **enforces** this demands extra persistent state (the current write position) to verify the next call to `psa_fwu_write()` which I am not so happy with.

athoelke, 2023-01-27:
Michel - I have created a PR to add this feature to the API for 1.0.0 final: #49. I would appreciate it if you could take a look at the changes.

jamike, 2023-05-23:
Andrew, 
After the last comment on #49, it seems that this ticket could be closed.  To check specification modification, before closing,
 I want to regenerate a specification in pdf format. WhereI can I find explanation for document regeneration ?

jamike, 2023-05-23:
Thanks , for the link to document,  I reviewed it and it's fine.
I also think that PSA_FWU_LOG2_WRITE_ALIGN can reduce error. But 2 power 0 should be possible for device without alignment constraints. I suggest alignment is 1<< PSA_FWU_LOG2_WRITE_ALIGN ?

athoelke, 2023-05-23:
@jamike - I have implemented a patch for this on PR #49, and included an updated PDF.

commit b56ea7e4 (2023-06-30): Add alignment requirement for image writes

Add PSA_FWU_WRITE_ALIGN to constrain the permitted image offsets, and permit implementation padding, when writing to a firmware image.

This enables efficient implementations on flash memories that have block-based write operations.

Fixes #5
closed 2023-06-30 by athoelke