GlobalPlatform/psa-api #8: Clarify expected behavior when a component has transient staging

athoelke, 2022-11-15:
The initial implementation of the 1.0 API in TF-M v1.7 will not support persistent staging. Images being prepared for update (i.e. components in WRITING or CANDIDATE state) are not preserved when the system restarts. Support for this is planned in a later version of TF-M

This issue was identified as part of the TF-M implementation of the 1.0-beta API. The implementation patch and review is: https://review.trustedfirmware.org/c/TF-M/trusted-firmware-m/+/17427.

Flexibility in the persistent/transient nature of the WRITING and CANDIDATE states is supported by the v1.0 API, but the specification provides no details on the expected impact on the state model, and implementation behavior.

### Proposal

Extend [Appendix C *Variation in system design parameters*](https://arm-software.github.io/psa-api/fwu/1.0/appendix/variations.html) to cover this use case.

Review other specification text to ensure that it is consistent with this use case.

athoelke, 2023-01-06:
I think I want to link this variation with support for the use case described in #9: they are both related to the bootloader only supporting a limited set of persistent states for the firmware store.

In the case of TF-M v1.7, mcuboot does not leave the firmware store in any state other than READY or TRIAL after a reboot - it will ignore/discard any partially staged image, and always cleans up the second image after an installation success or failure, unless it is preserved for a rollback-after-trial.

## Proposal

I propose to make the `PSA_FWU_FLAG_VOLATILE_STAGING` flag indicate that all of the affected states (WRITING, CANDIDATE, FAILED, UPDATED) are transient; i.e. they are not present immediately after a reboot. The FWU implementation should support these states during a single power cycle, i.e. the component state is maintained in the service memory, but not the persistent firmware store.

This has the caveats described in #9 - the Update Client will need to use other mechanisms to determine if an update was succesful, or failed, over a system restart.
Are there use

athoelke, 2023-01-06:
It would be useful to also clarify the effect of the `PSA_FWU_FLAG_VOLATILE_STAGING` on components that do not require a reboot.

My initial thoughts are that this behaves the same as with components that install over reboot: states other than READY are only maintained during the current power cycle, and the component will be in a READY state after a reboot.

This behavior is obvious for components that also do not use a TRIAL state (see the [modified state flow](https://arm-software.github.io/psa-api/fwu/1.0/appendix/variations.html#components-that-require-neither-a-reboot-nor-a-trial)):

* If the component state was WRITING, CANDIDATE, or FAILED prior to reboot, it will revert to READY, as if `cancel` and `clean` transitions had occurred.
* If the component state was UPDATED prior to reboot, it progresses to READY, as if a `clean` transition had occurred.

If the component has a TRIAL state, it is not clear what should happen at reboot if the component is in TRIAL state. For this type of component (see the [state flow](https://arm-software.github.io/psa-api/fwu/1.0/appendix/variations.html#components-that-require-a-trial-but-no-reboot)), there is no obvious automatic trigger to cause a rollback of the component in the normal flow. This is noted in the text.

**Question:** Should a system restart cause such a TRIAL to be rolled back automatically? - and should that always be the case for such a component, or only if the component has volatile staging?

Sherryzhang2, 2023-02-01:
In MCUboot, a trial image will be reverted automatically in booting up.

athoelke, 2023-02-02:
@Sherryzhang2 - does TF-M support firmware update for components which are updated without a reboot? - the question in my last comment was specifically about components that do not require a reboot to complete installation.

The reason the issue arises only for that kind of component, is that transition from CANDIDATE to TRIAL occurs without a reboot, so it is not self-evident that a reboot should cause the TRIAL to end, or that the bootloader would have any responsibility for managing the installation state of a such a component.

However, without a specified point at which a TRIAL must be terminated, the state model (see [here](https://arm-software.github.io/psa-api/fwu/1.0/appendix/variations.html#components-that-require-a-trial-but-no-reboot)) can be left in TRIAL state indefinitely, providing a rollback-attack facility via the `psa_fwu_reject()` function, and also blocking the ability for the API to start any new update of any other component.

I am inclined to add some text to _recommend_ that the implementation should reject such a TRIAL during a system restart/reboot, if it has not been explicitly accepted or rejected before then, as a mitigation against this threat. In practice this might occur as part of the Update Service or platform initialisation, rather than in the device bootloader.

PR #48 (2023-04-20): Clarify the effect of volatile staging
Add state diagrams to show the effect of volatile staging, and improve the description of the state model, and its variation.

Draft PDF preview: [IHI0093-PSA_Certified_Firmware_Update_API-1.0.0-pr48.pdf](https://github.com/ARM-software/psa-api/files/10969947/IHI0093-PSA_Certified_Firmware_Update_API-1.0.0-pr48.pdf)

Fixes #8
Fixes #9
closed 2023-04-20 by athoelke