GlobalPlatform/psa-api #14: New algorithm: Zigbee's block-cipher based hash (AES-MMO)

athoelke, 2022-11-29:
Zigbee defines a cryptographic hash based on a block-cipher using the Matyas-Meyer-Oseas (MMO) construction. See [*Zigbee Specification r21*](https://zigbeealliance.org/wp-content/uploads/2019/12/docs-05-3474-21-0csg-zigbee-specification.pdf) §B.6.

The MMO construction is general, requiring:
* a block cipher that has a key length equal to the block size
* a padding operation on the input message to align with the block size
* an IV (or salt) of block-size length

Zigbee specifies all of these details for its hash function based on MMO:
* AES-128 is used as the block cipher
* The padding operation is similar to that used for MD and SHA hashes to prevent length-extension attacks, and permitting messages of less than 2<sup>32</sup> bits in length
* The IV is set to 0 (all bits zero)

The existing Crypto API for hash algorithms is not parameterized or salted. So supporting this use case as a Crypto API hash algorithm requires a new hash algorithm identifier for the Zigbee-specified hash function.

athoelke, 2022-11-29:
This hash algorithm would naturally lead to the associated HMAC algorithm, which is also required/specified by the Zigbee specification §3.1.4.

athoelke, 2023-07-06:
### API Proposal

My suggestion is to call this hash function `PSA_ALG_AES_MMO_ZIGBEE` - to indicate it is based on the MMO 'mode'/construction over AES-128, and is specific to the length padding and parameterization specified by Zigbee.

We could use any unallocated hash algorithm identifier value. Is there any reason that `0x02000006` is unsuitable?

The Zigbee keyed hash function would not need a custom identifier, it is just `PSA_ALG_HMAC(PSA_ALG_AES_MMO_ZIGBEE)`.

gilles-peskine-arm, 2023-07-07:
> Is there any reason that 0x02000006 is unsuitable?

I can't think of any. The gap between SHA-1 and SHA-224 is so that the 4-element SHA2 family starts at a multiple of 4.

PR #94 (2024-01-17): Add support for zigbee algorithms
Add support for algorithms required by zigbee:
* MMO hash construction over AES-128
* CCM* optionally-authenticated variant of CCM

This PR adds a new hash function, and an unauthenticated cipher for the zero-length tag variant of CCM*. For non-zero-length tags, the existing support for CCM is adequate.

Fixes #14
Fixes #15
closed 2024-01-17 by athoelke