GlobalPlatform/psa-api #15: New algorithm: Zigbee's encryption and authentication block-cipher mode (CCM*)

athoelke, 2022-11-29:
Zigbee defines a block-cipher mode, CCM*, that is derived from the CCM mode used for authenticated encryption with additional data (AEAD). See [*Zigbee Specification r21*](https://zigbeealliance.org/wp-content/uploads/2019/12/docs-05-3474-21-0csg-zigbee-specification.pdf) §B.1.1 and §A.

CCM is already supported by the Crypto API: see [`PSA_ALG_CCM`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/aead.html#c.PSA_ALG_CCM).

CCM* differs from CCM in that it additionally supports a construction of CCM which only provides encryption, i.e. authentication tag length *M* = 0, while retaining the CCM construction of the encryption stream from the nonce, configuration parameter *L*, and CTR mode.

Zigbee constrains the use of CCM* as follows:
* the block-cipher is AES-128
* the message-length encoding parameter, *L* = 2, limiting the plaintext length to < 2<sup>16</sup>
* the tag length *M* must be 0, 4, 8 or 16

Further, the construction of the 13 byte nonce is specified precisely, with the final byte encoding the value of *M* that is used (as part of the packet's 'security level'). See §4.5.2.2 in the Zigbee specification.

Zigbee's packet security can also optionally encrypt the payload. This is achieved by presenting the payload as part of the additional data for CCM* when encryption is not indicated by the security level, or as the plaintext for CCM* when encryption is required. See §4.4.1.1 and §4.4.1.2.

athoelke, 2022-11-29:
There are some complications in adding CCM*, or Zigbee's constrained CCM*, into the Crypto API.

Although CCM* generally fits into the AEAD algorithm category:
* when the tag length *M* > 0, CCM* is identical to CCM
* when the tag-length is zero, this is actually just a stream cipher that requires a nonce (not a random IV) for security

## Option 1: Implement CCM*[*M*=0] as a Crypto API cipher algorithm

Issues to resolve:

- The Crypto API for unauthenticated ciphers does not support an application-provided IV in the single-part functions. Secure use of this encryption mode, and Zigbee applications in particular, would be required to use the multi-part API in order to use `psa_cipher_set_iv()` to set the nonce for encryption or decryption.

- In CCM, the value of *L* (which specifies how the message length is encoded), is not provided explicitly by the algorithm, but is deduced from the length of the nonce (if provided), or the plaintext length. In a CCM*[*M*=0] cipher, *L* would have to be deduced from a provided IV length. This reinforces the previous issue. Limiting the cipher algorithm to the Zigbee selected *L*=2 would mitigate this issue, but constrain the flexibility of the API definition.

- _Based on my understanding of the specification_, a Zigbee application can use the same key to secure packets at different security levels. Security levels without authentication would use the CCM*[*M*=0] cipher algorithm, security levels with authentication would use the CCM AEAD algorithm. Either the Crypto API would need to allow a single key to be used with both these algorithms, or the application would have to maintain two copies of the key with different permitted algorithm policies.

## Option 2: Implement CCM* as a Crypto API AEAD algorithm

This approach immediately fixes the primary issues with using a cipher algorithm, as the API requires a nonce to be specified (or generated), with a reasonable approach to deducing *L* (inherited from CCM), and permits a single key to be used across the entire family of tag lengths.

However, it raises a different set of concerns:

- A zero length tag does not provide authentication - thus CCM*[*M*=0] is **not** an AEAD algorithm. This loss of security will be significantly less evident in code using CCM* than if the `psa_cipher_xxx()` APIs are used.

- To help separate use of CCM from the maybe-not-authenticated CCM*, a separate set of algorithm identifiers could be used. Is it confusing to have different identifiers for the same algorithm (when the tag length is non-zero)?

- It would be 'neat' to allocate `PSA_ALG_AEAD_WITH_SHORTENED_TAG(PSA_ALG_CCM, 0)` as the AEAD algorithm for CCM* with no tag. (At present the L field of an AEAD algorithm identifier is not permitted to be zero). However, this would make CCM* and CCM indistinguishable within an implementation of the API. The risk of this exposing CCM-using applications to abuse, is somewhat mitigated by the need for the permitted algorithm policy for a key to explicitly permit short tags

## Option 3: Implement a Zigbee-specific flavor of 'AEAD' algorithm

Instead of generic CCM*, provide a Zigbee-specific AEAD algorithm that fixes the parameters (*L* = 2, *M* = 0,4,8, or 16). This algorithm does permit no authentication.

The relationship with CCM is not exposed in the algorithm identifier definition, and a single key cannot be used with both CCM and Zigbee AEAD algorithms.

The separation of algorithm identifiers removes the risk of using this algorithm when it should not be used. It can also enable the implementation to verify/enforce the relationship between the value of *M* and the security level encoded in the nonce.

However, this approach goes against the general spirit of the Crypto API to provide crypto building-blocks for applications/protocols, rather than higher-level functionality. On the other hand, the Crypto API provides protocol-specific support for KDFs that are only used in TLS, for example.

athoelke, 2022-11-29:
> CCM* without a tag is not authenticated, so it must not be exposed through the AEAD API.

Does Zigbee use the same key with both `PSA_ALG_CCM_STAR_NO_TAG` and `PSA_ALG_CCM`? - or is the security level for a single key instance never changed?

If the former, should there be a key usage policy that permits a single key to be used with both of these algorithms? - or is it an acceptable security risk to make the application manage the key material outside of the key store for this?

athoelke, 2022-12-09:
Given the existing definition of `PSA_ALG_CCM_STAR_NO_TAG` as a Crypto API cipher in Mbed-TLS, and with good arguments against doing anything different, we need to consider the application usage issue that this presents.

I've done more reading to better understand the Zigbee security architecture. Zigbee has adopted AES-CCM* as it provides the ability for optional encryption and optional authentication with the same key, without compromising the key. Zigbee's use of CCM* also prevents attacks which attempt to adjust the security level of data packets. In particular, the security level (use encryption and length of authentication tag) is encoded in the nonce/iv used with the CCM* operation.

This is unlike the IEEE 802.15.4 MAC security (which Zigbee runs on) which uses AES-CBC, AES-CTR, or AES-CCM to provide a spectrum of security support. Using the same key with more than one of these modes carries risks due to the way the input data is transformed by these algorithms.

Based on reading so far, I _think_ that the security level in a Zigbee network is set once by a policy in the initial Coordinator, and not changed after that. So, it _might_ be possible for a Zigbee implementation to use the Crypto API with just the `PSA_ALG_CCM_STAR_NO_TAG` definition: depending on the security level, the keys can be created or generated with an appropriate choice of `PSA_ALG_CCM_STAR_NO_TAG` or `PSA_ALG_AEAD_WITH_SHORTENED_TAG(PSA_ALG_CCM, MIC_length)` as the permitted algorithm policy.

However, the design of CCM* is specifically to permit the same key to be safely used with varying levels of authentication and encryption; so long as this is encoded in the nonce that is used. It is reasonable for an application, that is using CCM*, to want a single key to be used for `PSA_ALG_CCM` (optionally with a shortened tag) and for `PSA_ALG_CCM_STAR_NO_TAG`.

I propose that we introduce a new algorithm identifier, that is only usable as a permitted algorithm policy, for applications that which to use a key for the full range of CCM* functionality.

My initial suggestion is `PSA_ALG_CCM_STAR_ANY_TAG`, and to allocate this as a wildcard AEAD algorithm distinct from the existing CCM algorithm identifiers.

As a result, it would be reasonable to define a key policy

gilles-peskine-arm, 2022-12-13:
> I propose that we introduce a new algorithm identifier, that is only usable as a permitted algorithm policy, for applications that which to use a key for the full range of CCM* functionality.
>
> My [...]

Agreed, except it should be a cipher wildcard, not AEAD, since you're only guaranteed confidentiality from it, not authentication.

athoelke, 2023-07-06:
## Draft API Proposal

Before drafting this as a set of updates to the text, I'd like to validate my understanding of the discussion so far. Is this the API we are expecting:

### `PSA_ALG_CCM_STAR_NO_TAG` (macro)

The CCM* cipher mode without authentication.

```
#define PSA_ALG_CCM_STAR_NO_TAG                 ((psa_algorithm_t)0x04c01300)
```

This is CCM* as specified in IEEE 802.15.4 §7, with a tag length of 0. For CCM* with a nonzero tag length, use the AEAD algorithm `PSA_ALG_CCM`.

The underlying block cipher is determined by the key type.

The IV generated or set in the cipher API is used as the nonce in the CCM* operation. An implementation must support the default IV length of 13. Support for setting a shorter IV is optional.

The maximum message length that can be encrypted is dependent on the length of the IV. See `PSA_ALG_CCM` for details of this relationship.

#### Usage

The Zigbee message encryption algorithm is based on CCM*:

*  For unauthenticated messages &mdash; when _M_ = 0 &mdash; the `PSA_ALG_CCM_STAR_NO_TAG` algorithm is used with an AES-128 key in a multi-part cipher operation. The 13-byte IV must be constructed as specified in [ZIGBEE], and provided to the operation using `psa_cipher_set_iv()`.

   An implementation of Zigbee cannot use the single-part cipher functions, as these generate a random IV, which is not valid for the Zigbee protocol.

*  For authenticated messages &mdash; when _M_ &isin; {4, 8, 16} &mdash; the `PSA_ALG_CCM` algorithm is used with an AES-128 key.

*  To enable a single AES-128 key to be used for both the `PSA_ALG_CCM_STAR_NO_TAG` cipher and `PSA_ALG_CCM` AEAD algorithm, the key can be defined with the wildcard `PSA_ALG_CCM_STAR_ANY_TAG` permitted algorithm.

#### Compatible key types

PSA_KEY_TYPE_AES
PSA_KEY_TYPE_ARIA
PSA_KEY_TYPE_CAMELLIA
PSA_KEY_TYPE_SM4

---

In the definition of `PSA_ALG_CCM`, I will add a note on its usage for implementing CCM* for Zigbee, and cross reference the usage note that is under `PSA_ALG_CCM_STAR_NO_TAG`.

---

### `PSA_ALG_CCM_STAR_ANY_TAG` (macro)

A wildcard algorithm that permits the use of the key with CCM* as both an AEAD and an unauthenticated cipher algorithm.

```
#define PSA_ALG_CCM_STAR_ANY_TAG                 ((psa_algorithm_t)0x04c09300)
```

|Rationale|
|-|
|This wildcard value is essentially `PSA_ALG_CCM_STAR_NO_TAG` with bit 15 set. This is the same bit that is used to signify a wildcard MAC algorithm identifier.|

A block-cipher key that specifies `PSA_ALG_CCM_STAR_ANY_TAG` as its permitted algorithm, can be used with both the `PSA_ALG_CCM_STAR_NO_TAG` unauthenticated cipher, and with the `PSA_ALG_CCM` AEAD algorithm.

PR #94 (2024-01-17): Add support for zigbee algorithms
Add support for algorithms required by zigbee:
* MMO hash construction over AES-128
* CCM* optionally-authenticated variant of CCM

This PR adds a new hash function, and an unauthenticated cipher for the zero-length tag variant of CCM*. For non-zero-length tags, the existing support for CCM is adequate.

Fixes #14
Fixes #15
closed 2024-01-17 by athoelke