GlobalPlatform/psa-api #16: Incremental/partial initialization of a Crypto implementation

athoelke, 2022-11-30:
/**
 * \brief Partial library initialization.
 *
 * Applications may call this function on the same subsystem more than once.
 * Once a call succeeds, subsequent calls with the same subsystem are
 * guaranteed to succeed.
 *
 * Initializing a subsystem may initialize other subsystems if the
 * implementations needs them internally. For example, in a typical
 * client-server implementation, #PSA_CRYPTO_SUBSYSTEM_COMMUNICATION is
 * required for all other subsystems, and therefore initializing any other
 * subsystem also initializes #PSA_CRYPTO_SUBSYSTEM_COMMUNICATION.
 *
 * Calling psa_crypto_init() is equivalent to calling
 * psa_crypto_init_subsystem() on all the available subsystems.
 *
 * \note This function is part of a draft API specification and may change
 *       as the API evolves.
 *
 * \note You can initialize multiple subsystems in the same call by passing
 *       a bitwise-or of \c PSA_CRYPTO_SUBSYSTEM_xxx values. This ability
 *       is experimental and may change without notice. If the initialization
 *       of one subsystem fails, it is unspecified whether other requested
 *       subsystems are initialized or not.
 *
 * \retval #PSA_SUCCESS
 * \retval #PSA_ERROR_INSUFFICIENT_MEMORY
 * \retval #PSA_ERROR_INSUFFICIENT_STORAGE
 * \retval #PSA_ERROR_COMMUNICATION_FAILURE
 * \retval #PSA_ERROR_HARDWARE_FAILURE
 * \retval #PSA_ERROR_CORRUPTION_DETECTED
 * \retval #PSA_ERROR_INSUFFICIENT_ENTROPY
 * \retval #PSA_ERROR_STORAGE_FAILURE
 * \retval #PSA_ERROR_DATA_INVALID
 * \retval #PSA_ERROR_DATA_CORRUPT
 */

## `psa_crypto_init_subsystem` (function)

    psa_status_t psa_crypto_init_subsystem(psa_crypto_subsystem_t subsystem);

athoelke, 2024-07-08:
Working through the proposal to construct a draft PR, I notice a handful of issues that are worth resolving:

1. The statements about what functionality does, and does not, required `PSA_CRYPTO_SUBSYSTEM_RANDOM` seem particularly MbedTLS-based (not surprising, this is where the above text was taken from):

   > Only the following operations are guaranteed not to require this subsystem:
   >
   > *   Hash operations;
   > *   Signature verification operations.
   >
   > Currently, symmetric decryption for ciphers and AEAD, and MAC operations do not require the random generator. This may change in future versions of the library or when the operations are performed by a driver.

   Is there anything we could or should state categorically about this subsystem for the API specification?

2. The definition of `psa_crypto_init_subsystem()` includes error codes that are not currently listed against `psa_crypto_init()` in the specification (although I see these are listed there in the mbedtls doxygen comments):

   *  `PSA_ERROR_INSUFFICIENT_STORAGE`
   *  `PSA_ERROR_HARDWARE_FAILURE`
   *  `PSA_ERROR_STORAGE_FAILURE`
   *  `PSA_ERROR_DATA_INVALID`
   *  `PSA_ERROR_DATA_CORRUPT`
   
   Are these all useful? - if so, I presume they should be added to `psa_crypto_init()` in the spec?

3. What behavior should be specified for calls to `psa_crypto_init_subsystem()` when the `subsystem` parameter includes bits that are not recognized/supported by the implementation? We could do either:

   a) Ignore unrecognized bits - in effect these unknown subsystems are 'already initialized', as they do not need further initialization.
   b) Report an INVALID_ARGUMENT or NOT_SUPPORTED error if unrecognized/unsupported bits are set.

   I think I prefer (a), as this fits a model of usage where a caller does not have to know implementation details (which subsystems are implemented) to use the API. But there might be a good argument for (b) that I haven't considered....

athoelke, 2024-07-08:
Almost all function calls can return `PSA_ERROR_BAD_STATE` with the description:

> The library requires initializing by a call to `psa_crypto_init()`.

With the addition of `psa_crypto_init_subsystem()`, should we:

1. Leave this text as is, even if a call to `psa_crypto_init_subsystem()` with a suitable parameter value would also have the required effect.
2. Generalize the statement to something like the following with a section reference:

   > The library requires initializing. See _Library initialization_.

3. Enhance each instance to describe both `psa_crypto_init()` and `psa_crypto_init_subsystem()` (without providing subsystem details - as those may be implementation defined).

My current thinking is to do (2).

athoelke, 2024-07-08:
> > The statements about what functionality does, and does not, required PSA_CRYPTO_SUBSYSTEM_RANDOM seem particularly MbedTLS-based
> 
> I think there are two plausible choices here. One is to make it entirely implementation-defined. The other is what I made a guarantee in Mbed TLS, which is that you can verify a signature before randomness is available. The second choice removes doubt that in a bootloader, you can verify the signature of the next bit (which may include drivers for other mechanisms).

I see that, and this is already reflected in the existing specification text [in the 'Note' here](https://arm-software.github.io/psa-api/crypto/1.2/api/library/library.html#c.psa_crypto_init) regarding initialization of an RNG. I think this use case is important enough to keep a specification-level requirement for hashes and signature verification not requiring the RANDOM subsystem.

> > What behavior should be specified for calls to psa_crypto_init_subsystem() when the subsystem parameter includes bits that are not recognized/supported by the implementation?

> ...
> (TL,DR: leaning towards b.)

Do we want applications that want to use this facility to need to write code like this:

```
psa_status_t r = psa_crypto_init_subsystem(PSA_CRYPTO_SUBSYSTEM_COMMUNICATION);
if (r != PSA_SUCCESS && r !=PSA_ERROR_INVALID_ARGUMENT && r !=PSA_ERROR_NOT_SUPPORTED) {
    return ERR_INIT_FAILED;
}
```

... just in case they are compiled against a client-server implementation, and a non-client-server implementation is permitted to report any of SUCCESS, INVALID_ARGUMENT, or NOT_SUPPORTED in this situation?

I think I am leaning toward an in-between approach:

* For defined (in the spec) subsystem values, the implementation returns SUCCESS if either (a) it has no such subsystem to initialize, or (b) it has successfully initialized that subsystem.
* For undefined subsystem values, the implementation should return INVALID_ARGUMENT.

> We need to describe somewhere what you can do with various subsystems initialized. I don't have a strong opinion as to where. To some extent, this will be implementation-defined: for example only algorithms that require randomness are guaranteed to fail if RANDOM isn't initialized (and even then, an implementation may initialize RANDOM implicitly, for example a client in a client-server implementation will typically have everything available as soon as COMMUNICATION is established).

I have been reworking the text that is currently in the description of `psa_crypto_init()` as an introduction to the _Library initialization_ section, but expanded to describe both one-shot and partial initialization. This previously warned that what can be done prior to calling `psa_crypto_init()` is implementation-defined. I am adjusting that to describe the implementation-defined nature of the relationship between subsystem initialization and functionality.

gilles-peskine-arm, 2024-07-08:
> * For defined (in the spec) subsystem values, the implementation returns SUCCESS if either (a) it has no such subsystem to initialize, or (b) it has successfully initialized that subsystem.
> * For undefined subsystem values, the implementation should return INVALID_ARGUMENT.

Yes, that was my intent. For example COMMUNICATION is effectively always enabled in a non-client-server implementation, and init(COMMUNICATION) should always succeed.