GlobalPlatform/psa-api #17: Clarify the requirements for the hash parameter in calls to `psa_sign_hash()` and `psa_verify_hash()`

athoelke, 2022-11-30:
The specification of `psa_sign_hash()` and `psa_verify_hash()` does not place requirements to check that the input hash is valid for the signature algorithm.

Partly, this is due to the use in some protocols of separating the 'hash' step from the signing step; and where the input to the signing step is not the output of a cryptographic hash operation.

We propose to tighten the specification of these functions, to recommend that the implementation makes more effort to validate the input for algorithms where this is meaningful:

* Ensure that the format of the `hash` parameter is defined for all algorithms that can be used with these functions
* Add '`hash` is not a valid input value for the signature algorithm `alg`' to the list of conditions that can result in a `PSA_ERROR_INVALID_ARGUMENT` return code for these functions
* Explicitly require that `hash_length` must be equal to the output length of the associated hash function, for signature algorithms where the input to the signature **is** a hash.

athoelke, 2022-12-19:
> For `PSA_ALG_RSA_PKCS1V15_SIGN_RAW`, should we require that the `hash` parameter is a correct DER encoding of a message digest in a _DigestInfo_ structure

I can answer my own question: the primary reason that this algorithm is included is to support implementations of TLS 1.0 and 1.1. In these protocols, the signature is made over the concatenation of the MD5 and SHA-1 hashes of the messages, and not following the standard RSA PKCS#1 v1.5 encoded _DigestInfo_.

So, for this algorithm, the specification cannot require strict validation of the input.

gilles-peskine-arm, 2022-12-19:
Not only TLS 1.0/1.1, but also `CKM_RSA_PKCS` in PKCS#11 (which is even more general; in PSA we only generalize for signature, not for other operations.).

PR #37 (2023-02-14): Clarify the hash parameter in hash signature functions
Fixes #17
commit e4228cda (2023-02-14): Clarify the hash parameter in hash signature functions

Fixes #17
closed 2023-02-14 by athoelke