GlobalPlatform/psa-api #45: Mismatch between Mbed TLS documentation and J-PAKE specification (RFC8236) in mbedTLS 3.3

oberon-sk, 2023-01-12:
The specification of `PSA_ALG_JPAKE` (in [_crypto_extra.h_](https://github.com/Mbed-TLS/mbedtls/blob/v3.3.0/include/psa/crypto_extra.h) as well as in _psa_crypto_api_pake_ext.pdf_) includes mandatory calls to [`psa_pake_set_user()`](https://github.com/Mbed-TLS/mbedtls/blob/v3.3.0/library/psa_crypto_pake.c#L321) and `psa_pake_set_peer()` to set the local and peer user ids. The implementation in mbedTLS 3.3, however, disallows calls to these functions (`PSA_ERROR_NOT_SUPPORTED`) and uses the default ids "client" and "server" instead. These ids are often used but are not mandatory according to the J-PAKE specification in [RFC8236](https://www.rfc-editor.org/rfc/rfc8236).

Is this a limitation of the current Mbed TLS implementation (and tests) or will the specification be changed accordingly?

mpg, 2023-01-24:
I believe the specification is correct: as you say, nothing in RFC 8236 mandates uses of those particular strings, so we should accept other IDs.

I'm afraid the discrepancy between the current Mbed TLS implementation (and test) and the specification wasn't intentional: I believe it happened naturally while trying to build the PSA implementation on top of the legacy API (`ecjpake.h`) which is actually quite specific to TLS, and it appears we failed to noticed while reviewing https://github.com/Mbed-TLS/mbedtls/pull/5871 - sorry for this.

This is a bug in Mbed TLS that will be fixed, I've just created https://github.com/Mbed-TLS/mbedtls/issues/6961 to track it.

Thanks for your report, and also sorry for not reacting to your previous report on the psa-crypto mailing list - I had tagged for further review, but somehow missed it until now.

oberon-sk, 2023-01-24:
Thanks @mpg, I am glad if we could help.

athoelke, 2023-08-08:
Resolved as an FAQ

closed 2023-08-08 by athoelke

mpg, 2023-08-09:
Note: this has been fixed in the latest version of Mbed TLS, which is now aligned on the PSA spec (which itself was always in agreement with RFC8236) see https://github.com/Mbed-TLS/mbedtls/issues/7149 and https://github.com/Mbed-TLS/mbedtls/issues/7289.