GlobalPlatform/psa-api #72: Release Candidate for Firmware Update API 1.0.0

athoelke, 2023-05-25:
**PDF for review**: [IHI0093-PSA_Certified_Firmware_Update_API-1.0.0-rc1.pdf](https://github.com/ARM-software/psa-api/files/11569993/IHI0093-PSA_Certified_Firmware_Update_API-1.0.0-rc1.pdf)

This includes all merged and proposed changes since 1.0-beta, including #48, #49, #68, and #71.

The change history in Appendix E provides a summary of the changes, and links to the primary sections that are impacted. For a detailed diff of the changes, see the _Files changed_ tab for the specific Pull Request.

Please provide feedback about anything that is unclear, confusing, missing, or incorrect; so we can finalize the document for publication.

stevew817, 2023-06-20:
Page 16: unclear sentence "Downloading and installing updates must be reliable to wasting energy on..."

athoelke, 2023-06-21:
> Page 16: unclear sentence "Downloading and installing updates must be reliable to wasting energy on..."

Thank you - a missing word, it was meant to say: "Downloading and installing updates must be reliable to __avoid__ wasting energy on..."

athoelke, 2023-07-25:
Release Candidate 2: There has been enough feedback from RC1 to justify publishing an updated Release Candidate for final review.

I've updated the issue description above with a new PDF for RC2, and links to the PRs and commits with the changes between RC1 and RC2.

athoelke, 2023-08-02:
RC2 review has concluded.

closed 2023-08-02 by athoelke