GlobalPlatform/psa-api #93: Update to the PAKE API - should this remain a Beta?

athoelke, 2023-08-08:
There are a number of issues related to the PAKE API at the moment. The addition of support for SPAKE2+ (see #73), and issues identified during implementation of the current Beta API (see #86, #87, #88 and #89).

The resolution of these issues and enhancements should be combined into an update to the PAKE API. I see a couple of ways to release those changes:

1. As another *Beta* of the PAKE Extension, on the assumption that we are uncertain that we have identified all of the API details for supporting this class of algorithm. In which case, we would aim to publish something like a "Crypto API 1.2 PAKE Extension Beta-1" document, as we are working on a 1.2 update to the main specification.

* Alternatively, this could be upgraded to *Final*, on the assumption that we have identified the major details of the API via multiple implementations, and supporting multiple PAKE algorithms; and also on the expectation that application usage of the API will increase due to the use of SPAKE2+ in Matter and ECJPAKE in TLS 1.2, and API stability becomes important. In this case, we would aim to integrate the PAKE API into the Crypto API 1.2 (or perhaps 1.3) release, as we would no longer require a separate document.

Is there a strong argument for choosing one of these approaches over the other?

silabs-Kusumit, 2023-09-01:
Just thinking out loud here, I think the document could remain in Beta until we have the implementation of SPAKE2+ and see if that raises any additional issue than the ones already identified. I think that we would have a strong case for finalizing the API after adding and reviewing the implementations of both augmented (SPAKE2+) and balanced (EC-JPAKE) PAKE algorithms.

athoelke, 2024-01-10:
> Just thinking out loud here, I think the document could remain in Beta until we have the implementation of SPAKE2+ and see if that raises any additional issue than the ones already identified. I [...]

This is now the official plan. I am preparing to issue a release candidate of the Beta-2 PAKE Extension, which will be published in due course. Once an implementation has validated that the API is good for these algorithms, the document will be re-published as a Final version.

PR #160 (2024-01-31): Change of PAKE extension status to REL/Final
Following review of a proof-of-concept implementation of the Beta-2 API specification, I would like to change the status of the PAKE API to 'final'/'release'. This will make the PAKE API subject to the same change control as the main Crypto API specification.

* Update the configuration for 'Final' status (RC2)
* Update references to the status
* Update description of the API status throughout the document
* Update the introduction text relating to the adoption of J-PAKE and SPAKE2+

Fixes #93
closed 2024-01-31 by athoelke