GlobalPlatform/psa-api #104: Add API for RNG re-seeding

ndevillard, 2023-10-06:
A common implementation for random number generators is to use a TRNG to provide an initial seed which is then used by a DRBG to generate pseudo-random bits. This is mainly due to the fact that TRNGs need some time to gather entropy, which may be too long a wait for some applications. Implementations reserve a protected area to store the DRBG seed and make sure it is periodically refreshed, e.g. by pulling bits from a TRNG if available, or at least hashing the existing seed to avoid repeating sequences.
Without making any assumptions on how the seed is initially populated or refreshed, an additional function for the API should be provided to explicitly trigger such a refresh: `psa_random_seed(void)`. This would enable:
- Devices with a fast enough TRNG, where `get_random()` returns TRNG-generated bits and `random_seed` does nothing.
- Devices with a slow TRNG, where `get_random()` returns DRBG-generated bits based on the current seed and `random_seed` uses TRNG outputs to re-seed the DRBG.

Proposal: add `psa_random_seed()` with IMPDEF behaviour. Provide documentation with recommendations for implementers.

gilles-peskine-arm, 2023-10-06:
Feeding data from a TRNG is already covered by the driver interface.

> an additional function for the API should be provided

Why should this be covered by the _application_ interface? The API is designed so that applications do not need to be trusted, and does not have a mechanism to designate certain applications as trusted. This can be added as an implementation-specific feature, but we have so far deliberately not made it a standard feature.

(Entropy injection from an untrusted source is harmless if you just mix it in, barring implementation vulnerabilities. But it's also useless: either the root of trust already has enough entropy and doesn't need this injection, or it doesn't have entropy and still won't have enough entropy after the injection.)

ndevillard, 2023-10-06:
Ah I got it completely wrong. Let me re-formulate the need into another ticket.

closed 2023-10-06 by ndevillard