GlobalPlatform/psa-api #106: Add support for CMAC based KDF algorithms

adeaarm, 2023-10-09:
Current revision of the spec does not have support for CMAC based KDFs as described by NIST in [SP 800-108](https://nvlpubs.nist.gov/nistpubs/legacy/sp/nistspecialpublication800-108.pdf) in the list of KDF algorithms, but it support only HMAC based algorithms. This is important for use cases where CMAC used as the PRF in the KDF is fully accelerator by the underlying hardware platform. PRF can be used in different modes, ideally add support for all of them to extend flexibility for platform integrators.

MarcusJGStreets, 2023-10-09:
There are issues around CMAC, raised within SP 800-108 with mitigations. So we would need to ensure that the API enforced those. 

When using CMAC as the PRF in counter mode, a party with knowledge of the key KIN, who
also can freely select portions of the string Label || 0x00 || Context || [L] 2 (as described in
Appendix B), may be able to control the values of one or more of the K(i). One way to guard
against the possibility of this sort of key control would be by replacing the fixed input data
Label || 0x00 || Context || [L] 2
with
Label || 0x00 || Context || [L] 2 || K (0),
where
K(0) = PRF(KIN, Label || 0x00 || Context || [L] 2).

When using CMAC as the PRF in feedback mode, a party with knowledge of the key KIN, who
can also freely select portions of the string Label || 0x00 || Context || [L] 2 and/or portions of the
IV, may be able to control the values of one or more of the K(i), as was the case for counter
mode. One way to guard against the possibility of this sort of key control would be by requiring
that
IV = PRF (KIN, Label || 0x00 || Context || [L]2 )
and requiring the inclusion of a counter in the feedback mode (i.e., [i]2 is not optional).

When using CMAC as the PRF in double-pipeline mode, a party with knowledge of the key KIN
may be able to control the values of one or more of the K(i). One way to guard against the
possibility of this sort of key control would be by requiring the inclusion of a counter in double-
pipeline mode (i.e., [i] 2 is not optional).

athoelke, 2023-10-18:
SP 800-108 provides general construction methods for building KDFs, but does not define any specific KDF construction, that is adequate for implementing an interoperable KDF. We want to only define algorithm identifiers for precise algorithm specifications, that ensure compatibility of results across implementations.

HKDF and the HMAC-based TLS KDFs are defined because their behavior is fully specified.

If you have specific CMAC-based KDFs, ideally with an example use case, then adding an algorithm identifier and describing the KDF will be straightforward. Clearly, CMAC-based KDFs that follow secure-construction guidelines are preferable to ones that do not.

gilles-peskine-arm, 2023-10-18:
[SP 800-108](https://nvlpubs.nist.gov/nistpubs/legacy/sp/nistspecialpublication800-108.pdf) unfortunately defines very general families that leave a lot of choices in terms of how you format the data to be processed.

* There are three different scheme.
* Each scheme takes multiple parameters.
* The discussion in §5 specifies that the pseudocode given in the next subsections does not fully define the functions: implementations are also allowed to change the order of the concatenation `[i] || Label || 0x00 || Context || [L]`.

There is a [draft NIST specification for a JSON syntax that describes the capabilities of an SP 800-108 implementation](https://pages.nist.gov/ACVP/draft-celi-acvp-kbkdf.html). Read and despair.

SP 800-108 KDF in counter mode is a generalization of ANSI X9.62 KDF1, which uses SHA-1 as the PRF. Limiting the API to **the structure of KDF1, but with HMAC/KMAC/CMAC rather than SHA-1 as the primitive**, might be viable.

[PKCS#11](https://docs.oasis-open.org/pkcs11/pkcs11-spec/v3.1/os/pkcs11-spec-v3.1-os.html#_Toc111203684) allows a lot of variation, and even there I'm not sure if it covers all the functions that are compliant to SP 800-108. The result is too complicated for PSA.

[OpenSSL 3.1](https://www.openssl.org/docs/man3.1/man3/) doesn't have KDFs other than the ones used by TLS. Neither does BouncyCastle, as far as I can find.

[Cryptodome](https://pycryptodome.readthedocs.io/en/latest/src/protocol/kdf.html#sp-800-180-counter-mode) only offers counter mode, with one specific encoding which is not specified in the documentation.

Just to add to the fun, [SP 800-56C §4](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-56Cr2.pdf) specifies a “one-step key derivation function” which follows the same principle as SP 800-108's KDF in counter mode, but formats the data in an incompatible way! Whereas SP 800-108 passes the secret input as a key for a key-indexed PRF, SP 800-56C passes the secret input as part of some data which is concatenated into an input to a single PRF. SP 800-56C allows a lot of variation over the FixedInfo part (i.e. encoding various non-secret parameters). [SP 800-56A §5.8.2.1](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-56Ar3.pdf) gives additional rules regarding the formation of FixedInfo in various use cases, but these rules are about non-secret inputs, so for our API design, we can let the application construct it. Note that SP 800-56C does not support CMAC (it needs a keyed hash, not just a MAC, because it uses a non-secret key; SP 800-56C over CMAC would be broken).

So I think we should specify the following NIST-approved KDF families:

* A subset of SP 800-108 in counter mode, based on HMAC, KMAC or CMAC. This subset should be as close as possible to KDF1 and to Cryptodome's `SP800_108_Counter`.
    * Note that the CMAC variant is limited to a secret input of size exactly 16, 24 or 32 bytes since it's directly used as an AES key. This makes the CMAC variant possibly vulnerable to related-key attacks in some scenarios.
* SP 800-56C in all generality, with hash, HMAC and KMAC variants.

*

adeaarm, 2023-10-19:
Just wanted to confirm that having point 1 (A subset of SP800-109 in counter mode, especially in the CMAC variant) would fulfill most of the requirements that we see from partners around CMAC based KDF. I think this should be due to the fact that it's common to have accelerators where the CMAC functionality can be accelerated entirely (and possibly hardened in HW against side channels).

athoelke, 2023-10-19:
Do partners know what the applications are, and can tell us precisely what KDF they need? - or are they expecting that if we specify some CMAC-based KDF, and they implement it, that it will get used by applications because it is accelerated?

athoelke, 2023-10-19:
> * A subset of SP 800-108 in counter mode, based on HMAC, KMAC or CMAC. This subset should be as close as possible to KDF1 and to Cryptodome's `SP800_108_Counter`.

It will be challenging* to achieve alignment with both KDF1 structure (i.e. replace the hash with a *MAC as the PRF) and Cryptodome `SP800_108_Counter()` (assuming it literally follows §4.1 in SP 800-108).

[* As in technically impossible]

If the HMAC/CMAC/KMAC operation is denoted as $\text{PRF}(s, x)$, without output length $prflen$, then the number of output blocks required is $n=\lceil L/prflen \rceil$ if $L$ is the number of bytes of output requested. I have used $[x]_n$ to denote an encoding of the integer $x$ as an $n$-byte string. Then the blocks of output for KDF1 are:

$$K_i = \text{PRF}(K_{IN}, [i]_4), \quad i=0..n-1$$

Whereas SP 800-108 §4.1 Counter mode outputs the following blocks:

$$K_i = \text{PRF}(K_{IN},[i]_4\text{ || }Label\text{ || }0x00\text{ || }Context\text{ || }[L]_8), \quad i=1..n$$

[I have adopted a 4-byte encoding of $i$ and an 8-byte encoding of *L* in this demonstration - it is not documented what CryptoDome does].

Even with empty $Label$ and $Context$ values, these are incompatible due to the zero-based and one-based indexing, and the additional fixed data in the PRF input.

The SP 800-108 construction is preferable as it provides placeholders for shared $Context$ information (identity/protocol etc.) that help to indicate protocol errors, and key separation via $Label$ for key schedules that do multiple derivations.

However, the inclusion of $L$ in the PRF input is problematic for the PSA Crypto key derivation API, as we do not have a one-shot key derivation function, and the application can continue to extract additional output. We could take one of the following approaches:

* Recognize that this is optional in the NIST paper, and just drop $[L]$ from the PRF input
* Used the fixed value $[0]_8$ instead of $[L]_8$
* If the application calls `psa_key_derivation_set_capacity(len)`, then use $[\mathtt{len}]_8$ in the PRF input, otherwise $[0]_8$

The last of these might permit interoperability with CryptoDome (assuming there is agreement on the PRF input structure), by setting the capacity before extracting any output.

adeaarm, 2023-10-19:
> Do partners know what the applications are, and can tell us precisely what KDF they need? - or are they expecting that if we specify some CMAC-based KDF, and they implement it, that it will get used by applications because it is accelerated?

A couple of examples are available [here](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/2.4.99-dev2/nrf/security/tfm.html#encrypted-its) and [here](https://git.trustedfirmware.org/TF-M/trusted-firmware-m.git/tree/platform/ext/accelerator/cc312/cc312-rom/cc3xx_kdf.h)

I believe in both cases, given that the underlying acceleration technology is similar, the driving requirement is the capability of having it fully accelerated in HW (that HW could be hardened against side channels), rather than some peculiarity of the chosen algorithms, i.e. it's a platform integration constraint. For this reason I believe that the expectation is to have the *some* algorithm based on CMAC for KDF, standardized in the spec. And once it is standardized, it will be used because it is accelerated on such platforms and for such use cases, at the same time keeping compliance to the PSA Crypto standard.

athoelke, 2023-10-20:
> For this reason I believe that the expectation is to have the _some_ algorithm based on CMAC for KDF, standardized in the spec. And once it is standardized, it will be used because it is accelerated [...]

It will be important to understand how much of the algorithm is accelerated (or 'difficult to change'). If it just the PRF (CMAC-AES) then what happens with the KDF structure around that is a matter of software/firmware/driver coding. If the structure of the KDF itself is baked into those implementations, then it might be valuable to see if the PSA specification can match those existing implementations.

adeaarm, 2023-10-20:
> It will be important to understand how much of the algorithm is accelerated (or 'difficult to change'). If it just the PRF (CMAC-AES) then what happens with the KDF structure around that is a matter [...]

Talking for these particular implementations, it's the AES-CMAC block being accelerated (i.e. the PRF only), not the construction around it. And the AES HW block in particular is hardened against side channels. I believe such configuration would be common especially in IoT or low power in general targeted accelerators where implementing in HW the whole construction would increase the complexity.

gilles-peskine-arm, 2023-10-20:
To clarify and expand on something that I wrote earlier: the KDF from SP 800-108 have a subtle limitation when used with CMAC. They require a secret input KIN of length exactly 16, 24 or 32 bytes. If KIN is not secret (which includes the case where it's a low-entropy, guessable secret), an adversary can construct label or context strings that produce partially known output. This often defeats the security of the protocol that relies on the KDF. In a nutshell, **having a fixed-length secret is crucial**.

This is specifically a problem with CMAC. With key derivation functions based on a hash or HMAC or KMAC, the differences between the inputs don't really matter: you can use a public input as the “secret” and a private input as the “label”, and that doesn't actually change the practical security properties. So there's a high risk that users will take a protocol that's designed for the most common hash-based KDF, and will not realize that by using CMAC as the building block instead of hash/HMAC the protocol is no longer secure.

[SP 800-108r1](https://nvlpubs.nist.gov/nistpubs/SpecialPublications/NIST.SP.800-108r1.pdf) warns about this in §4.1 (bottom of page 7):

> When using CMAC as the PRF in counter mode, a party with knowledge of the key KIN, who also can freely select portions of the string Label || 0x00 || Context || [L]2 (as described in Appendix B), [...]

We can specify this construction instead of the default SP 800-108 one. But if we do that, we lose interoperability with common implementations. For example, I don't think you can build that on top of the PKCS#11 `CKM_SP800_108_COUNTER_KDF`. And if we don't have good interoperability anway, we might as well specify a different CMAC-based construction that can be implemented more efficiently. The SP 800-108 key derivation functions re-process the whole fixed label+context for each output block, which is costly both in processor time and RAM. If we have to build our own, we might as well select a more efficient construction.

athoelke, 2023-10-20:
> To clarify and expand on something that I wrote earlier: the KDF from SP 800-108 have a subtle limitation when used with CMAC. They require a secret input KIN of length exactly 16, 24 or 32 bytes. [...]

For a CMAC-based KDF we can mandate that the `PSA_KEY_DERIVATION_INPUT_SECRET` step must be a block-cipher key [that is compatible with CMAC] provided using `psa_key_derivation_input_key()`. This would naturally block a trivial application change from switching from an HMAC-based KDF to a CMAC-based KDF.  [In passing, I note that although HKDF-EXPAND uses the secret input as the HMAC key, we failed to specify that the secret key input to this algorithm should have type `PSA_KEY_TYPE_HMAC` when introducing this algorithm.]

athoelke, 2023-10-20:
> We can specify this construction instead of the default SP 800-108 one. But if we do that, we lose interoperability with common implementations. For example, I don't think you can build that on top [...]

I wondered about leaving this mitigation step to the application, for when the key is known to an attacker: if we specified that the application can specify the entire 'Info' data instead of Label and Context. However, the application would need to use $K_{IN}$ for a CMAC computation, as well as the KDF, which isn't supported by our single-permitted-algorithm policies. If we think hardening the CMAC case by default is better, then I agree that the algorithm definition doesn't need to directly align with what we do for SP800-108-HMAC.

> The SP 800-108 key derivation functions re-process the whole fixed label+context for each output block, which is costly both in processor time and RAM. If we have to build our own, we might as well select a more efficient construction.

All of the modes in SP 800-108 process the fixed info once for each output block. HKDF-Expand does the same (but uses a feedback-with-counter style of construction). I'm not sure we should start designing our own CMAC-based KDF, especially if the CMAC computation is borne by hardware.

However, we could simplify the input structure, but include the general design and known-key mitigation from SP 800-108. A suggestion:

Inputs:
* $K_{IN}$: a CMAC key
* $Info$: a possibly-empty application-selected string
* $L$: length of output required

Process:

* $K_0=\mathrm{PRF}(K_{IN},Info)$
* $K_i=\mathrm{PRF}(K_{IN},K_{i-1}\text{ || }[i]_4\text{ || }Info), \quad i=1, ..., n$

Output:

* The first $L$ bytes of $K_1\text{ || }K_2\text{ || }...\text{ || }K_n$

Instead of feedback-mode, could just use $K_0$ in all of the following calls to the PRF, as recommended in the mitigation for CMAC in SP 800-108 counter mode.

Vge0rge, 2023-10-27:
> Do partners know what the applications are, and can tell us precisely what KDF they need? - or are they expecting that if we specify some CMAC-based KDF, and they implement it, that it will get used by applications because it is accelerated?

I work for Nordic Semiconductor and I just want to back what @adeaarm  said already. The most important thing for us it to have the CMAC AES based PRF defined in the PSA APIs. AES CMAC is hardware accelerated and we use it for key derivation. 

Having said that and following the very interesting discussion here I think that following a NIST recommendation (even if it leaves some things unclear) is for me an easier decision than implementing our our CMAC PRF variant.  Enforcing the NIST mitigation inside the APIs is not a problem.

@athoelke  Just to make sure that I follow you correctly here. Is your suggestion to allow the users to do a single call to the key derivation APIs where the input whatever structure they want? So they will have to prepare a buffer which (if they want to follow the NIST recommendation) will be: 

Label || 0x00 || Context || [L]2

and the driver will have to prefix the counter and suffix the K0?

athoelke, 2023-10-27:
> Is your suggestion to allow the users to do a single call to the key derivation APIs where the input whatever structure they want? So they will have to prepare a buffer which (if they want to follow [...]

Basically, yes. As the Crypto API permits incremental output of bytes and/or keys, the key derivation operation does not necessarily know how much output it should produce. In contrast, SP 800-108 constructions show the inclusion of [L] in the PRF input, to diversify the output depending on how much derived material is needed.

My suggestion is that instead of an application providing separate *Label*, *Context*, and *L* parameters to the operation, it just supplies a single *Info* parameter. If the application wants this to match a specific SP 800-108-based design, it needs to structure this *Info* buffer appropriately.

But as Gilles highlighted, if we adopt the SP 800-108r1 recommendation for CMAC, and add the $K_0$ suffix to the PRF input, then the KDF we specify will not be interoperable with CryptoDome or with PKCS#11 implementations. So perhaps interoperability with other crypto libraries should not be a priority for this function.

athoelke, 2023-10-27:
### Proposal

So a more concrete proposal for consideration:

Add two new algorithms:
* `PSA_ALG_SP800_108_COUNTER_HMAC()`, using HMAC as the PRF, parameterised by a hash algorithm
* `PSA_ALG_SP800_108_COUNTER_CMAC`, using CMAC as the PRF.

Both of these algorithms have two parameters:

* A required secret input key, $K_{IN}$, that must be an appropriate key for the PRF. The key's permitted algorithm must be the KDF. This key is input using:

      psa_key_derivation_input_key(&kdf, PSA_KEY_DERIVATION_INPUT_SECRET, key);

* An optional $Info$ buffer parameter that is supplied using:

      psa_key_derivation_input_bytes(&kdf, PSA_KEY_DERIVATION_INPUT_INFO, &info, sizeof(info));

  If no `PSA_KEY_DERIVATION_INPUT_INFO` parameter is provided, then $Info$ is the zero-length string.

The HMAC KDF is constructed as per SP 800-108r1 §4.1, but using $Info$ in place of $Label\text{ || }0x00\text{ || }Context\text{ || }[L]_2$ as the fixed data input to the PRF. Note that this enables an application to construct the $Info$ parameter to match other implementations of the NIST recommendation. The iteration counter is encoded in 32 bits, so that the construction does not unduly limit the output of the key derivation operation. So the output stream $K_1\text{ || }K_2\text{ || }K_3\text{ || }...$ is computed as:

$$K_i = \mathrm{HMAC}(K_{IN}, [i]_4\text{ || }Info), \quad i = 1, 2, 3, ...$$

The CMAC KDF is constructed as per SP 800-108r1 §4.1, including the recommendation to make the design robust against attackers with knowledge of $K_{IN}$, and also using $Info$ in place of $Label\text{ || }0x00\text{ || }Context\text{ || }[L]_2$ as the fixed data input to the PRF. The iteration counter is encoded in 32 bits, so that the construction does not unduly limit the output of the key derivation operation. So the output stream $K_1\text{ || }K_2\text{ || }K_3\text{ || }...$ is computed as:

$$K_0 =\mathrm{CMAC}(K_{IN}, Info)$$

$$K_i = \mathrm{CMAC}(K_{IN}, [i]_4\text{ || }Info\text{ || }K_0), \quad i = 1, 2, 3, ...$$

### Alternatives

We could alternatively, or in addition, support the use of `PSA_KEY_DERIVATION_INPUT_LABEL` and `PSA_KEY_DERIVATION_INPUT_CONTEXT`, and use these to construct the fixed input data as described in SP 800-108. We would still need to decide on whether, and how, to handle the $[L]$ content of this data. However, the simple definition with just the $Info$ parameter provides sufficient application flexibility and security.

Vge0rge, 2023-10-29:
Thank you for the detailed proposal @athoelke, this clarifies things a lot.

I think that an issue that may arise with this recommendation it is more error prone to errors from the users who need to prepare the buffer themselves and I don't see clearly the advantages of it since we already think that we will not have interoperability with other major implementations (in order to follow safer guidelines)

I see from the PSA API that the the set_capacity can be mandatory for some algorithms:

> Optionally set the operation’s maximum capacity with [psa_key_derivation_set_capacity()](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_set_capacity). This can be done before, in the middle of, or after providing inputs. For some algorithms, this step is mandatory because the output depends on the maximum capacity.

So what forbids us from using the capacity as the L value? We can make this call mandatory for the CMAC PRF since the algorithm is using it anyway.

Then we can rely on the existing `PSA_KEY_DERIVATION_INPUT_LABEL` and `PSA_KEY_DERIVATION_INPUT_CONTEXT` to provide the rest of the data and we can describe in the algorithm description in PSA that the counter and the L as 32-bit length values. 

This way we can reduce the effort from the users to do the buffer preparation and we make it clear for other implementations which may want to be compliant with PSA later what we do with this.

athoelke, 2023-10-30:
Part of my proposal was to support an HMAC-based Counter-mode KDF, which could be interoperable with other crypto libraries, if they directly follow the SP 800-108 recommendations, so long as there is agreement on the bit-size and encoding of the iterator and the output length. My suggestion was a compromise that increased the chance of interoperability for the HMAC based KDFs.

I would also be happy to have the API expect optional *Label* and *Context* inputs, and construct the fixed data from these. However, if we are deprioritizing interoperability, then I do not see a strong case for including *L* in the input data, which requires additional effort from the application when using the API?

Vge0rge, 2023-10-31:
I understand the argument for HMAC-based algorithm.
I think that the main argument for keeping the L argument for the CMAC at least is to be the closest that we can to the NIST recommendation. 
Do you think that it is sensible to allow two modes of operation?

Let's say:  `PSA_KEY_DERIVATION_MODE_RAW_INPUT` and  `PSA_KEY_DERIVATION_MODE_NIST_INPUT`. (The names are not great but hopefully good enough for the demonstration only).

Then for both the algorithms:
```
 PSA_ALG_SP800_108_COUNTER_HMAC(PSA_KEY_DERIVATION_MODE_RAW_INPUT)
 PSA_ALG_SP800_108_COUNTER_CMAC(PSA_KEY_DERIVATION_MODE_RAW_INPUT)
```
 We say that the only assumption that we do in this mode is that the counter is 32-bits and the fixed input need to be provided by doing a single call with the `PSA_KEY_DERIVATION_INPUT_INFO` from the user. (So any other input should not be allowed)

Then we don't add the mitigation with the K0 and we made it possible to have interoperability for implementations for both CMAC and HMAC.

Then we have the variant with the NIST recommended Input:
```
 PSA_ALG_SP800_108_COUNTER_HMAC(PSA_KEY_DERIVATION_MODE_NIST_INPUT)
 PSA_ALG_SP800_108_COUNTER_CMAC(PSA_KEY_DERIVATION_MODE_NIST_INPUT)
```

There we support the inputs that we need using the `PSA_KEY_DERIVATION_INPUT_LABEL` and `PSA_KEY_DERIVATION_INPUT_CONTEXT` and a call to the set capacity and we document that for the HMAC version we do:
Label || 0x00 || Context || [L]2

and for the CMAC version we do:
Label || 0x00 || Context || [L]2 || K0

This makes the algorithms a bit more complicated but if we document them enough I don't see that it can cause many issues.

athoelke, 2023-10-31:
Thanks for the suggestion @Vge0rge, we have concurrently thought of a different way to solve the issue of 'what should L be if the application does not call `psa_key_derivation_set_capacity()?` - we specify that for these algorithms, the default capacity is 2<sup>29</sup>-1, or `0xfffffff1` bytes. This would make $L = 2^{32}-8$ bits, the largest number of octets that can be encoded in a 32-bit $L$ value. If the application adjusts the capacity before output of data, this value is used instead to compute $[L]_4$ in the fixed input.

Then we can use the SP 800-108 structure, with $Label$ and $Context$ inputs, a 32-bit big-endian encoding for the iterator $i$, and also for the output length $L$.

It does not seem to be a priority, at the moment, to ensure compatibility with other crypto libraries/APIs for these functions - just between PSA Crypto API implementations. So we could leave a more flexible algorithm specification ('RAW_INPUT') until such interoperability use cases come to light.

Vge0rge, 2023-11-06:
This sounds a very good suggestion to me @athoelke :+1:

athoelke, 2023-11-06:
> This sounds a very good suggestion to me @athoelke 👍

Thanks for the feedback. I will update the proposal to verify understanding/agreement, and plan to work on a PR.

athoelke, 2023-11-06:
### Updated Proposal

Add two new algorithms:
* `PSA_ALG_SP800_108_COUNTER_HMAC()`, using HMAC as the PRF, parameterised by a hash algorithm
* `PSA_ALG_SP800_108_COUNTER_CMAC`, using CMAC as the PRF.

Both of these algorithms have one required, and two optional parameters:

* A required secret input key, $K_{IN}$, that must be an appropriate key for the PRF. The key's permitted algorithm must be the KDF. This key is input using:

      psa_key_derivation_input_key(&kdf, PSA_KEY_DERIVATION_INPUT_SECRET, key);

* Optional $Label$ and $Context$ buffer parameters can be supplied using:

      psa_key_derivation_input_bytes(&kdf, PSA_KEY_DERIVATION_INPUT_LABEL, &label, sizeof(label));
      psa_key_derivation_input_bytes(&kdf, PSA_KEY_DERIVATION_INPUT_CONTEXT, &context, sizeof(context));

  If a parameter is not provided, then it is treated as a zero-length string.

For these algorithms, the initial capacity of the key derivation operation is $2^{29}-1$ bytes (`0xfffffff1u`). The capacity can be explicitly set to a lower value by calling `psa_key_derivation_set_capacity()`.

When the first output is requested, the value $L$ is calculated as `8 * psa_key_derivation_get_capacity()`, and used in the subsequent derivation process. Subsequent calls to `psa_key_derivation_set_capacity()` have no effect on $L$.

* The HMAC KDF is constructed as per SP 800-108r1 §4.1. The iteration counter and output-length are encoded as big-endian 32 bit values, so that the construction does not unduly limit the output of the key derivation operation. The output stream $K_1\text{ || }K_2\text{ || }K_3\text{ || }...$ is computed as:

  $$K_i = \mathrm{HMAC}(K_{IN}, [i]_4\text{ || }Label\text{ || }\texttt{0x00}\text{ || }Context\text{ || }[L]_4 ), \quad i = 1, 2, 3, ...$$

* The CMAC KDF is constructed as per SP 800-108r1 §4.1, including the recommendation to make the design robust against attackers with knowledge of $K_{IN}$. The iteration counter and output-length are encoded as big-endian 32 bit values, so that the construction does not unduly limit the output of the key derivation operation. The output stream $K_1\text{ || }K_2\text{ || }K_3\text{ || }...$ is computed as:

  $$K_0 =\mathrm{CMAC}(K_{IN}, Label\text{ || }\texttt{0x00}\text{ || }Context\text{ || }[L]_4)$$

  $$K_i = \mathrm{CMAC}(K_{IN}, [i]_4\text{ || }Label\text{ || }\texttt{0x00}\text{ || }Context\text{ || }[L]_4\text{ || }K_0), \quad i = 1, 2, 3, ...$$

Vge0rge, 2023-11-07:
Thank you very much for the updated proposal @athoelke, your effort is much appreciated. This latest proposal seems very clear to me, it fits our use cases and I believe that it promotes a safe usage of the NIST proposal.

athoelke, 2023-11-07:
> When the first output is requested, the value $L$ is calculated as `8 * psa_key_derivation_get_capacity()`, and used in the subsequent derivation process. Subsequent calls to `psa_key_derivation_set_capacity()` have no effect on $L$.

For these algorithms, perhaps it would be better to prohibit calls to `psa_key_derivation_set_capacity()` after output has commenced. The principle being something like: "Operation capacity cannot be changed after beginning output for KDF algorithms where the capacity is used as one of the inputs"?

PR #123 (2023-12-11): Add support for KDFs based on SP 800-108 recommendations
Define algorithm identifiers for HMAC and CMAC variants of a counter-mode KDF that follows the construction recommended by NIST SP 800-108r1.

As the specification defines a general construction, the Crypto API specification provides a precise definition, to ensure compatibility between different implementations. See #106 for the enhancement request, and discussion.

Fixes #106
closed 2023-12-11 by athoelke