GlobalPlatform/psa-api #113: Should psa_key_derivation_verify_bytes be allowed with a USAGE_DERIVE secret?

gilles-peskine-arm, 2023-10-30:
[`psa_key_derivation_input_key`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_input_key) on a secret input (`PSA_KEY_DERIVATION_INPUT_SECRET` or `PSA_KEY_DERIVATION_INPUT_PASSWORD`) that is passed as a key allows the following operations:

* `output_xxx` ([`psa_key_derivation_output_bytes()`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_output_bytes), [`psa_key_derivation_output_key()`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_output_key)) only if the secret key has the policy [`PSA_KEY_USAGE_DERIVE`](https://arm-software.github.io/psa-api/crypto/1.1/api/keys/policy.html#c.PSA_KEY_USAGE_DERIVE).
* `verify_xxx` ([`psa_key_derivation_verify_bytes()`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_verify_bytes), [`psa_key_derivation_verify_key()`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/kdf.html#c.psa_key_derivation_verify_key)) only if the secret key has the policy [PSA_KEY_USAGE_VERIFY_DERIVATION](https://arm-software.github.io/psa-api/crypto/1.1/api/keys/policy.html#c.PSA_KEY_USAGE_VERIFY_DERIVATION).

You can't use a verify function on a key with only the DERIVE usage. This is similar to signature keys not automatically allowing verification. However, I'm not sure that's desirable. For backward compatibility, you might have a key created against the 1.0 API (which only had `USAGE_DERIVE`) and want to use it with the 1.1 API's `verify_xxx` functions.

Allowing DERIVE keys to be used as a secret for `verify_bytes` wouldn't change the security of the key, since you can simulate it with `output_bytes` followed by `memcmp`. Allowing DERIVE keys to be used as a secret for `verify_key` changes what you can effectively do with the keys involved, but I can't think of a scenario where this change would be problematic.

Are there good hygiene reasons to forbid `verify_bytes` and `verify_key` with a secret input that only allows `USAGE_DERIVE`?

gilles-peskine-arm, 2023-11-02:
As there are no objections, we'll allow `verify_bytes` and `verify_key` with a `DERIVE`-usage-only secret. We'll make that change in 1.2 since it's an incompatible change for 1.1-compliant implementations.

athoelke, 2023-11-02:
Should the same policy relaxation be applied to the 'key to match' that is passed to `psa_key_derivation_verify_key()`, as well as to the key provided as the SECRET or PASSWORD input to the KDF?

The historic usage argument is weak here, as there was no previous way to compare the result of a KDF with a stored PASSWORD_HASH key, without exporting the key data. So there is unlikely to be existing keys that have a well defined 'wrong policy' for this use case?

gilles-peskine-arm, 2023-11-02:
The key-to-match has a fundamentally different role and should generally not itself be used to derive further keys. So no, the key to match should not have `USAGE_DERIVE`.

It's in fact weird that the same policy flag is used for inputs to a KDF usable for verification, and for inputs to an equality comparison.

PR #118 (2023-11-13): KDF relaxations and clarifications

*   Changed the policy for `psa_key_derivation_verify_bytes()` and `psa_key_derivation_verify_key()`, so that these functions are also permitted when an input key has the `PSA_KEY_USAGE_DERIVE` usage flag.
*   Removed the special treatment of `PSA_ERROR_INVALID_SIGNATURE` for key derivation operations. A verification failure in `psa_key_derivation_verify_bytes()` and `psa_key_derivation_verify_key()` now puts the operation into an error state.
*   Clarified the behavior of a key derivation operation when there is insufficient capacity for a call to `psa_key_ [... cut]
closed 2023-11-13 by athoelke