GlobalPlatform/psa-api #141: Crypto PAKE Extension beta.2 : Review of Release Candidate

athoelke, 2024-01-10:
PDF for review: [AES0058-PSA_Certified_Crypto_API-1.2_PAKE_Extension-bet.2-rc1.pdf](https://github.com/ARM-software/psa-api/files/13890452/AES0058-PSA_Certified_Crypto_API-1.2_PAKE_Extension-bet.2-rc1.pdf)

Beta.2 adds support for SPAKE2+, and updates the API following feedback from implementation of the original PAKE interface. See the related issues and PRs in the [Beta.2 Milestone](https://github.com/ARM-software/psa-api/milestone/8?closed=1).

Please provide feedback about anything that is unclear, confusing, missing, or incorrect; so we can finalize the document for publication.

athoelke, 2024-01-30:
New release candidate RC2, includes feedback from RC1 (#155, #156, and #159), and updates the API status to Final (#160).

athoelke, 2024-01-31:
> New release candidate RC2, includes feedback from RC1 (#155, #156, and #159), and updates the API status to Final (#160).

Processing error - I forgot to include #155 and #156 into the RC2 build. RC3 now available.

oberon-sk, 2024-02-02:
RC3 looks good to us, thanks!

oberon-sk, 2024-02-02:
Thanks Andrew,
that‘s really awesome!
Best 
Stephan

PR #168 (2024-02-13): Publish the Crypto 1.2 and PAKE Extension specifications
This also ends the review of the release candidates: closes #152 and closes #141
closed 2024-02-13 by athoelke