GlobalPlatform/psa-api #179: Add support for the SRP-6/6a PAKE protocol

athoelke, 2024-03-05:
SRP-6 is the currently most widely deployed augmented PAKE algorithm, based on a key exchange based on finite-field Diffie-Hellmann arithmetic.

An early version is defined in RFC 2945, the SRP-6 definition adds protection against additional attacks, and RFC 5054 defines how SRP-6 can be used with TLS.

However, there is no single definitive reference for SRP-6, and differences exist in both the construction of the key confirmation messages, and the processing of the raw DH output value into a shared secret. For inclusion into the Crypto API, a precise definition of the algorithm (or algorithms) will need to be provided.

athoelke, 2024-03-05:
An initial proposal for the API elements has been proposed in #178

stevew817, 2025-01-06:
Adding an upvote for SRP-6(a) as in use by the Homekit Accessory Protocol. An example implementation under Apache 2.0 exists e.g. [here (Python)](https://github.com/jlusiardi/homekit_python/blob/b7d9fe730159ec90bf7a82010150d38112d4b1fc/homekit/crypto/srp.py) or [here (Ruby)](https://github.com/karlentwistle/ruby_home-srp).