GlobalPlatform/psa-api #181: crypto: problematic specification of IV for PSA_ALG_CTR

gilles-peskine-arm, 2024-03-06:
The specification of [`PSA_ALG_CTR`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/ciphers.html#c.PSA_ALG_CBC_NO_PADDING) has inconsistent provisions regarding the counter block.

CTR mode does not have a single specification. It is really a family of modes that arrange for the counter value uniqueness in different ways.

> During the counter block update operation, the counter block is treated as a single big-endian encoded integer and the update operation increments this integer by 1.

This is one way to do it, and it's what Mbed TLS does.

> A counter block value must only be used once across all messages encrypted using the same key value. This is typically achieved by splitting the counter block into a nonce, which is unique among all message encrypted with the key, and a counter which is incremented for each block of a message.

This is not how it works: it's only compatible with the paragraph cited above if the none part is 0 bytes, but then the nonce part is not a nonce since reuse is allowed (as long as the resulting counter value is unique across blocks).

> For example, when using AES-CTR encryption, which uses a 16-byte block, the application can provide a 12-byte nonce when setting the IV. This leaves 4 bytes for the counter, allowing up to 2^32 blocks (64GB) of message data to be encrypted in each message.

This example does not comply to the specification above. Note that the length passed to `set_iv` is irrelevant to the internal operation of the cipher.

Also:

> A call to [`psa_cipher_set_iv()`](https://arm-software.github.io/psa-api/crypto/1.1/api/ops/ciphers.html#c.psa_cipher_set_iv) on a multi-part cipher operation requires an IV that is between 1 and n bytes in length, where n is the cipher block length. The counter block is initialized using the IV, and padded with zero bytes up to the block length.

We might as well allow a 0-byte IV to start the counter at 0.

athoelke, 2024-03-13:
It might be best to rework the text from the perspective of the application developer, and provide the information about the IV behavior is response to use cases:
* If the application is implementation a protocol with a specified nonce construction for this mode, then use `psa_cipher_set_iv()`.
* Otherwise, ideally, the application can construct a guaranteed-to-be-unique nonce (of 8-14 bytes, say). Then the application should use `psa_cipher_set_iv()`; and also ensure that messages do not exceed the maximum length to prevent a collision in the counter block.
* If unique nonces cannot be provided by the application, then use a random 16-byte IV.

gilles-peskine-arm, 2024-03-14:
When a key is single-use, an all-bits-zero IV is as good as any, and it can be conveyed as `{iv=NULL; iv_length=0;}.

On the other hand, passing an empty IV is a plausible mistake in the case of a multiple-use key. (“I don't know what to put here. Does NULL work? Yes, sweet.”) So I suppose we should keep rejecting that. If you do want an all-bits-zero IV for a single-use key, you have to write it explicitly like `{IV=""; iv_length=1}` and that is a more explicit decision.

So let's keep rejecting a 0-length IV.

athoelke, 2024-03-14:
> On the other hand, passing an empty IV is a plausible mistake in the case of a multiple-use key. (“I don't know what to put here. Does NULL work? Yes, sweet.”) So I suppose we should keep rejecting [...]

That's a good argument for requiring the application to explicitly select a constant (zero) IV, rather than accidently do so.

PR #186 (2024-03-22): Improve the description of using IVs in PSA_ALG_CTR
Fixes #181

[As we are patching 1.2 to fix some minor issues, it is worth including this clarification before 1.3.]
closed 2024-03-22 by athoelke