GlobalPlatform/psa-api #203: Dragonfly PAKE for WPA3 use cases

athoelke, 2024-07-17:
WPA3, part of the IEEE 802.11 wireless specifications, defines a key establishment mechanism called *Simultaneous Authentication of Equals* (SAE). The key exchange mechanism is a variant of the Dragonfly PAKE (see [RFC 7664](https://www.rfc-editor.org/rfc/rfc7664.html)).

The SAE protocol has evolved:
* On first debut in 802.11s (2011) for mesh networks, SAE used a 'hunting-and-pecking' (HNP) method for computing a group element (ECC or FF) that is very similar to the one described in RFC 7664.
* SAE was adopted for WPA3-Personal in 802.11-2016, to replace the use of WPA2 (and WEP).
* Vulnerabilities in the WPA3-SAE protocol, particularly related to the password to group element derivation, and the scope for down-grade attacks, were published in 2019.
* Countermeasures are implemented in the 802.11-2020 specification, introducing a preferred 'hash-to-curve' (aka hash-to-element or H2E) method for the group element computation that can be implemented in constant time, and adding information to the key derivation context to mitigate a downgrade attack.

SAE is fully specified in IEE 802.11-2020 §12.4, including the H2E and HNP methods, the key exchange, and the specific hash and key derivation procedures for the protocol.

athoelke, 2024-07-17:
If WPA3-SAE, or a more generic Dragonfly PAKE, is something you want to have added to the API, please respond here, so we can determine the importance of adding this protocol to the specification.

stevew817, 2025-01-06:
Upvoting on behalf of Silicon Labs.