GlobalPlatform/psa-api #239: API guarantees for output buffer length in multi-part AEAD

Damian-Nordic, 2025-02-05:
For a single-part AEAD operation, such as `psa_aead_encrypt`,  it is advised to use `PSA_AEAD_ENCRYPT_OUTPUT_SIZE(...)` macro to calculate the output buffer length that is guaranteed to be sufficient to hold the ciphertext and the tag. The macro typically returns the sum of the plaintext length and the tag length.

For a multi-part AEAD operation, such as `psa_aead_update`, there is a similar macro available, `PSA_AEAD_UPDATE_OUTPUT_SIZE(...)`, which calculates a sufficient buffer length to hold the ciphertext. The macro typically returns the plaintext length (passed in the current step) rounded up to a multiple of the cipher block length.

The problem is that because of the current definition of `PSA_AEAD_UPDATE_OUTPUT_SIZE(...)`, I am not sure if I can safely perform AES-CCM encryption/decryption in place, even if the plaintext length is not a multiple of the block length (I cannot use a single-part operation as in my case the tag is stored separately from the ciphertext). I wonder if one of the following changes could be done in the API documentation to guarantee in-place operations using multi-part AES-CCM:
1. The rounding to a multiple of the block size should not be needed for CCM as the encryption component of CCM behaves like a stream cipher.
2. There could be another macro that calculates the total output length that is sufficient to hold all the outputs from a sequence of `psa_aead_update` operations and the ending `psa_aead_finish` operation. Thus, even  `psa_aead_update` might defer (a part) of the output to a subsequent call, there would be a well-defined limit for the total output length.

athoelke, 2025-02-21:
The current wording in the specification is designed to support the use of hardware implementations which buffer inputs so that they process and output whole AES-blocks at a time. Although for such an implementation, the output will never overtake the input, the `PSA_AEAD_UPDATE_OUTPUT_SIZE()` macro operates without the context of the current operation state - again this is by design so that this macro can evaluate to a compile-time constant if the parameters are also compile-time constants.

MbedTLS does this, as it can be configured to use one or more hardware components. At least one of the hardware crypto drivers that it works with has the buffering behavior, even for the stream-cipher component of AES-CCM.

So, I do not think we can constrain the specification to guarantee the behavior suggested in option 1. I.e. for stream-based encryption algorithms, we cannot require the `xxx_SIZE()` macro to return the precise number, without block-rounding.

Unfortunately, even if the caller knows that the cipher-text output cannot overtake the input, the specification states that the value of output buffer memory that is between reported output length and the provided buffer size is 'unspecified' on success. This provision is to permit an implementation to clear the unwritten parts of output buffers, which can reduce the risk of vulnerabilities caused by the caller incorrectly processing residual data in the buffer. So, in practice, using the AEAD operation for an in-place encryption using non-block-aligned input buffers can result in input data being overwritten before it is used.

So option 2 would not actually resolve this use case either, without requiring that output buffer bytes not used for the actual output must remain unmodified by a call to the API. This may be more convenient, but is likely to lead to vulnerabilities in application code.

That said, a specific implementation, that does not provide any buffering of input or output for AES-CCM, could make the `PSA_AEAD_UPDATE_OUTPUT_SIZE()` macro report the exact output size (== input). Application code that relies on this behavior to do in-place encryption/decryption would be technically non-portable (as some implementations do not have this property), but that might not be a problem for your use case.

Alternatively, the application could break the input into block-aligned pieces, which (I think) will work correctly for all the implementations that I am aware of.

Damian-Nordic, 2025-02-21:
@athoelke Thanks a lot for the justification.

In the code we're working on, this makes us do an extra memory allocation or split `psa_aead_update` into two calls (so that a small stack buffer is used to encrypt the last `input_size % block_size` bytes). I was wondering whether the current wording was just an oversight or it had a reason, but I now understand it did.

Closing the issue then.

gilles-peskine-arm, 2025-02-21:
Here's Mbed TLS (and TF-PSA-Crypto)'s position on this topic. As Andrew explained, we can't make the output size macros tighter because this is impossible with some hardware implementations.

We could, however, make in-place sequential processing work, in scenarios such as the one described here. One way that comes to mind is to guarantee that certain functions do not affect the output buffer beyond the length of the actual output. But it is currently not the case and would require nontrivial engineering.

The difficulty is that many functions in our implementation “clean up” the whole output buffer, even beyond the length of the actual output. This is a general strategy to avoid leaving “bad” data around, e.g. to ensure that no sensitive content remains in the unused part of the buffer. Generally, we use this strategy in functions with “small” output, e.g. in a MAC or key-export function, we prefer to zero out the output buffer before doing anything else. Ideally we would always refrain from doing this in functions with “large” output. There's no fundamental obstacle there, “just” engineering time.