GlobalPlatform/psa-api #277: Update LAMPS-SHBS citations, and finalize LMS, HMS and XMSS key formats.

athoelke, 2025-06-27:
IETF has now published RFC 9802 (https://www.rfc-editor.org/info/rfc9802).

MarcusJGStreets, 2025-07-02:
I do not think we want to touch the ASN.1 within PSA. 
Leave interpreting that to higher layers. 
But do want to support the various options.

athoelke, 2025-07-02:
> I do not think we want to touch the ASN.1 within PSA.

Agreed. In this case, the RFC just wraps the LMS/HSS/XMSS public key data structures in a BIT STRING or OCTET STRING within the SubjectPublicKeyInfo. So I think we can keep the current PSA definitions for the public keys as being those data structures.

We do not define a private key format (but neither do any of the RFCs). These are stateful hashes, and export/import of private keys risks OTS reuse/abuse.