GlobalPlatform/psa-api #279: API to query a key's capabilities

gilles-peskine-arm, 2025-07-01:
### User stories

As a developer implementing a protocol involving cryptography (e.g. TLS),
I want to query the capabilities of a PSA key object,
so that I can decide which protocol options (e.g. cipher suites or key exchange types) the key is compatible with.

As a developer implementing a key management interface,
I want to query the capabilities of a PSA key object,
so that I can enable the correct operations on it.

### Requirement

In principle, both user stories can be implemented on top of `psa_get_key_attributes`: just analyze the key type and policy. However, this is difficult for several reasons:

* I need to have logic covering all the cryptographic mechanisms that I support. In the protocol user story, that's doable, but duplicates code that already exists in the PSA crypto implementation to determine whether an operation is possible. In the key management user story, I would have to reimplement everything.
* If the implementation supports vendor-specific mechanisms, they should be included. Again, that's necessary in the key management user story, and less burdensome in the protocol user story.
* If the implementation has additional policy restrictions, they should apply. For example, if an implementation allows small RSA keys to be used to verify (presumably old) signatures but not encrypt new material, that should be reflected in this capability interface.
* If the implementation has additional policies (such as Mbed TLS's enrollment algorithm, which allows a key to have a second permitted algorithm), this should be reflected.

### Mbed TLS

In Mbed TLS ≥3.2, this API exists for key pair objects under the name [`mbedtls_pk_can_do_ext()`](https://github.com/Mbed-TLS/mbedtls/blob/mbedtls-3.6.4/include/mbedtls/pk.h#L459).

```
int mbedtls_pk_can_do_ext(
    const mbedtls_pk_context *ctx,
    psa_algorithm_t alg,
    psa_key_usage_t usage);
```

* `ctx` is a proprietary representation of a key in memory. The PSA analog is a key identifier.
* `alg` is a PSA algorithm `PSA_ALG_xxx`.
* `usage` is a single `PSA_KEY_USAGE_xxx` flag.
* The return value is a boolean.

For example `mbedtls_pk_can_do_ext(ctx, PSA_ALG_RSA_PSS, PSA_KEY_USAGE_SIGN_HASH(PSA_ALG_SHA_256))` checks whether `ctx` is an RSA key which is allowed by policy to perform a PSS signature using SHA-256. It is always false for an ECC key regardless of policy.

In the [next version](), we want to [extend this function to also support public keys](https://github.com/Mbed-TLS/TF-PSA-Crypto/blob/6558a69b1e3e23af409f6cb7f2987dd65f3f40c1/docs/architecture/pk-4.md#new-semantics-of-mbedtls_pk_can_do_ext). So for example `mbedtls_pk_can_do_ext(ctx, PSA_ALG_RSA_PSS, PSA_KEY_USAGE_SIGN_HASH(PSA_ALG_SHA_256))` can only be true for an RSA key pair, whereas `mbedtls_pk_can_do_ext(ctx, PSA_ALG_RSA_PSS, PSA_KEY_USAGE_VERIFY_HASH(PSA_ALG_SHA_256))` can also be true for an RSA public key.

### Proposed interface

I think the design of the Mbed TLS function has proved itself, so I just propose to tweak it to fit PSA. Name TBD.

```
psa_status_t psa_check_key_usage(
    psa_key_id_t key,
    psa_algorithm_t alg,
    psa_key_usage_t usage);
```

* `key`: a PSA key identifier.
* `alg`: an algorithm. (Not a wildcard algorithm policy — I'm not sure whether this should mean “can do all of” or “can do at least one of”, and I propose to just reject wildcards for now.)
* `usage`: a single `PSA_KEY_USAGE_xxx` flag. Only flags corresponding to an operation that uses an algorithm are meaningful here; flags like `COPY` and `EXPORT` are not. (Those are simple booleans that you can query from the attributes, unlike operation capabilities which require checks on the key type and algorithm as well.)
* Return values:
    * `PSA_SUCCESS` — yes, the key can do this.
    * `PSA_ERROR_INVALID_ARGUMENT` or `PSA_ERROR_NOT_SUPPORTED` or `PSA_ERROR_NOT_PERMITTED` — no, the key can't do this.
    * `PSA_ERROR_INVALID_HANDLE` — the key doesn't exist.

### A flag for the public side of a key agreement

How do I ask whether a key is suitable for the public side of a key agreement? E.g. how do I formulate
```
psa_check_key_usage(key, PSA_ALG_ECDH, ???)
```
so that it's true for an ECC public key? Note that this cannot be `PSA_KEY_USAGE_DERIVE`, since
```
psa_check_key_usage(key, PSA_ALG_ECDH, PSA_KEY_USAGE_DERIVE)
```
requires a key pair.

This hasn't come up in key policies because `psa_key_agreement()` and its variants take the representation of the public key as an argument, rather than a key object. (The API is optimized for ephemeral Diffie-Hellman over the much rarer static Diffie-Hellman.)

Tentative proposal: a new usage flag `PSA_KEY_USAGE_DERIVE_PUBLIC`. What would it mean in key policies?

MarcusJGStreets, 2025-07-01:
Whatever we decide, this is going to be an edge case, due to the slight lack of symmetry in the API. 

My first feeling is that we should document PSA_KEY_USAGE_DERIVE_PUBLIC as only for use in this new function.

It should not be an error to add it to a key policy, but it has no effect.

I would also not have the function check that the key object has the policy, only check that the object is either a public key or a key pair - as the public key needs to be exported and is always exportable.

So in this case, it only checks that the key is the correct type and the algorithm is supported. That is, you will never get PSA_ERROR_NOT_PERMITTED if asking about using a public key in a Derive operation. Though, you would get PSA_ERROR_INVALID_ARGUMENT is the key is not a public key, and PSA_ERROR_NOT_SUPPORTED, if the implementation does nto offer the algorithm. 

But if you ask the same question of a private key, or ask about encryption or verification, you can get PSA_ERROR_NOT_PERMITTED.

gilles-peskine-arm, 2025-07-01:
Thinking further, it might make sense for the PSA API to query not only the possibility of doing a crypto operation, but the possibility of doing it in a specific way — one-shot, multipart, interrutible... I'm not convinced this is really useful, though, because the only use cases for it that I can think of are about runtime discovery of an implementation's feature support, rather than information about a specific key. And PSA is geared towards small systems where we expect feature discovery to be mostly settled at compile time. Also, in the cases where multipart/interruptible capabilities are useful, they could be reported by location and operation, rather than by key.

For this, we could add a `flags` parameter. And that parameter would give us extra room to express things that the usage can't encode, for example
```
psa_check_key_usage(key, PSA_ALG_ECDH, PSA_KEY_USAGE_DERIVE. PSA_CHECK_KEY_PRIVATE);
psa_check_key_usage(key, PSA_ALG_ECDH, PSA_KEY_USAGE_DERIVE. PSA_CHECK_KEY_PUBLIC);
```

I'm putting this on the table, but I don't like it very much. The multipart/interruptible discovery feels like it's taking the wrong argument and the public/private usage distinction feels like it's shoving into something unrealted.

athoelke, 2025-07-02:
There is one scenario in the existing API where a public-key-specific derivation usage flag might have been viable already. In the SPAKE2+ asymmetric PAKE, the Verifier can be provided with a 'public' key that is derived from the authentication password (which is not required to be confidential, as it does not enable the spoofing of the Prover).

In the spec, https://arm-software.github.io/psa-api/crypto/1.3/api/ops/pake.html#id2 describes how the Prover PAKE operation must be set up with a key-pair (the private key is the verifier key), but the Verifier PAKE operation can be provided either the key-pair or the public key.

For asymmetric PAKE, there might have been value in having a secondary DERIVE usage flag for keys that are permitted for use in the secondary/verifier role in an asymmetric PAKE. But we didn't consider this at the time, and the PAKE operation requires that it is set up with a key that has DERIVE usage.

This also hits the same issue that a `psa_check_key_usage()` on a SPAKE2+ verifier key for the SPAKE2+ algorithm cannot distinguish between the two roles, as the usage flag is the same for both roles.

athoelke, 2025-07-21:
The current favourite approach for this API design challenge is:

1. Introduce a new usage flag `PSA_KEY_USAGE_DERIVE_PUBLIC`.
2. This **would not** be used as part of a key policy usage flags value - it is not checked when calling any of the APIs to carry out a cryptographic operation. The correct key policy value for public key objects (e.g. used in SPAKE2+ verification) is still `PSA_KEY_USAGE_DERIVE`.
3. The new usage flag is **only** valid in calls to the new `psa_check_key_usage()` function, where is indicates the role being played by the key in the specified algorithm. For example:
   * `PSA_KEY_USAGE_DERIVE` with `PSA_ALG_ECDH` will check that the key can be used as the private share in the ECDH key agreement.
   * `PSA_KEY_USAGE_DERIVE_PUBLIC` with `PSA_ALG_ECDH` will check that the key can be used as the public share in the ECDH key agreement.
   * `PSA_KEY_USAGE_DERIVE` with `PSA_ALG_SPAKE2P_HMAC()` will check that the key can be used in the Prover role in the SPAKE2+ algorithm.
   * `PSA_KEY_USAGE_DERIVE_PUBLIC` with `PSA_ALG_SPAKE2P_HMAC()` will check that the key can be used in the Verifier role in the SPAKE2+ algorithm.
   * `PSA_KEY_USAGE_DERIVE_PUBLIC` with `PSA_ALG_HKDF()` is invalid, as there is no such role in single-key derivation algorithms.